enjoyneering/AHTxx #5: [AHT10@esp32 in PlatformIO] Error: Bus is in Slave Mode

bato3, 2022-07-14:
Default example not working.  I confirmed that the sensor works on esp8266 and esp32 with older version. `enjoyneering/AHT10`
```
[    16][E][Wire.cpp:283] setClock(): Bus is in Slave Mode
[   116][E][Wire.cpp:308] beginTransmission(): Bus is in Slave Mode
[   116][E][Wire.cpp:331] endTransmission(): Bus is in Slave Mode
AHT1x not connected or fail to load calibration coefficient
```

Behavior when no sensor on the I2C bus is connected, after power on. Messages do not appear if I disconnect the sensor and reset the DevBoard. (Without disconnecting the USB power)
```
Reconnecting to COM3 ...         Connected!
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4

[    16][E][esp32-hal-i2c-slave.c:577] i2c_slave_check_line_state(): Bus Invalid State, Can't init sda=0, scl=0
[    16][E][esp32-hal-i2c-slave.c:309] i2cSlaveInit(): bad pin state
[    21][E][Wire.cpp:167] begin(): Slave Init ERROR
[    25][E][esp32-hal-i2c.c:254] i2cSetClock(): could not acquire lock
[   132][E][esp32-hal-i2c.c:142] i2cWrite(): could not acquire lock
AHT1x not connected or fail to load calibration coefficient
```

pio run
```
PLATFORM: Espressif 32 (5.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3

Dependency Graph
|-- AHTxx @ 1.1.5+sha.d3de55e
|   |-- Wire @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
```

bato3, 2022-07-14:
I did a little more research in the code and it could be data type related, It's framework bug https://github.com/espressif/arduino-esp32/issues/6616

SDA and SCL are defined as 
```cpp
static const uint8_t SDA = 21;
static const uint8_t SCL = 22;
```
and `Wire.begin()` can be:
```cpp
    bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
    bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);
```

[Working solutions](https://github.com/enjoyneering/AHTxx/blob/main/src/AHTxx.cpp#L98):
```cpp
  Wire.begin((int) sda, (int)scl);
  Wire.begin(21, 22);
  Wire.begin();
```
Not working solutions:
```cpp
#define CONFIG_DISABLE_HAL_LOCKS true
Wire.begin(_address, (int)sda, (int)scl);
aht10.begin(21, 22)
```

And when I'm use `aht10.begin(-1, -1)`
```
[    15][E][esp32-hal-i2c-slave.c:224] i2cSlaveInit(): invalid pins sda=-1, scl=22
```
after change to `Wire.begin(_address, sda, scl);`
```
[    15][E][esp32-hal-i2c-slave.c:224] i2cSlaveInit(): invalid pins sda=-1, scl=-1
```

enjoyneering, 2022-07-16:
Fixed. Could you updated the library to 1.1.6 and let me know?

bato3, 2022-07-18:
It looks  OK / works

OT: why you don't choose solution with cast? Eg:

```cpp
Wire.begin(static_cast<int>(SDA), static_cast<int>(SCL));
```

closed 2022-07-18 by bato3

bato3, 2022-07-18:
BTW:

is still in `1.1.5`: https://github.com/enjoyneering/AHTxx/blob/main/library.properties#L2

https://github.com/enjoyneering/AHTxx/blob/main/src/AHTxx.h#L31
should be: `ESP8266, ESP-01` (missing coma)