enjoyneering/AHTxx #10: [PlatformIO registry] I can't find the library using PlatformIO registry search

hpsaturn, 2022-10-15:
## Summary

Hi, maybe the library was not published yet? because I don't found it via `pio` search tools (web and via CLI)

For example:

```bash
pio pkg search AHT
Found 1 packages (page 1 of 1)

adafruit/Adafruit AHTX0
Library • 2.0.3 • Published on Thu Sep 15 18:14:46 2022
```

enjoyneering, 2022-10-17:
it is not published. please add it manually. see [here](https://github.com/enjoyneering/AHTxx/issues/3).

closed 2022-10-17 by enjoyneering

hpsaturn, 2022-10-17:
Thanks for your response. It isn't a issue, is a feature request. We can use the library with specific commit, but we could lost possible new versions. And also for continuous integration pipe this change could improve the tests in projects that using the library. On the other hand the registry of PlatformIO is a good alternative to have more feedback or code contributions in your library. I don't know, maybe you should keep in mind it.