enjoyneering/AHTxx #11: M5StickCPlus sensors conflicts with last AHTxx version. All reads are blocked

hpsaturn, 2022-10-17:
## Summary

I'm using this library in my project with a wrapper that I wrote, CanAirIO Sensors Library. With the old version, AHT10 library and using other sensors at the same time, it was working fine in this device. With the new version, AHTxx library, it blocks the other sensors reads. In this wrapper we have charged multiple sensors libraries and we try to auto-recognize all because we have different devices and sensor configurations. In the previous version, when I had AHT10 initialization without any AHT sensor connected, it was working fine and don't block the other sensors.

More details please review [here](https://github.com/kike-canaries/canairio_sensorlib/issues/152)

enjoyneering, 2022-10-17:
I'll take a look. thank you.

enjoyneering, 2022-10-17:
Hi hpsaturn,

Try to change clock stretch and let me know?

```c++
void setup()
{
  Serial.begin(115200);
  Serial.println();
  
  while (aht10.begin(21, 22, 100000, 50000) != true)  //21=SDA, 22=SCL, 100000=i2c speed 100KHz, 50000=clock stretch 50ms
  {
    Serial.println(F("AHT1x not connected or fail to load calibration coefficient")); //(F()) save string to flash & keeps dynamic memory free

    delay(5000);
  }

  Serial.println(F("AHT10 OK"));

 //Wire.setClock(400000); //experimental I2C speed! 400KHz, default 100KHz
}
```

enjoyneering, 2022-10-17:
Also use the macro in AHTXX_USE_READ_DATA on the second read. This saves time as the temperature reading is already in the library buffer.

```c++
void Sensors::aht10Read() {
    float humi1 = aht10.readHumidity(AHTXX_FORCE_READ_DATA); //read 6-bytes via I2C, takes 80 milliseconds
    float temp1 = aht10.readTemperature(AHTXX_USE_READ_DATA); //AHTXX_USE_READ_DATA=use 6-bytes from humidity reading, takes zero milliseconds!!!
    if (humi1 != 255) humi = humi1;
    if (temp1 != 255) {
        temp = temp1-toffset;
        dataReady = true;
        DEBUG("-->[SLIB] AHT10 read\t\t: done!");
        unitRegister(UNIT::TEMP);
        unitRegister(UNIT::HUM);
    }
}
```

hpsaturn, 2022-10-18:
Hi

Thanks for your help, is working now using these implementations:

```cpp
void Sensors::aht10Init() {
    sensorAnnounce(SENSORS::SAHTXX);
    aht10 = AHTxx(AHTXX_ADDRESS_X38, AHT1x_SENSOR);
    #ifdef M5STICKCPLUS // issue: https://github.com/enjoyneering/AHTxx/issues/11
    if(!aht10.begin(32, 33, 100000, 50000)) return;
    #else
    if (!aht10.begin()) return; 
    #endif
    sensorRegister(SENSORS::SAHTXX);
}

void Sensors::aht10Read() {
    float temp1 = aht10.readTemperature(AHTXX_USE_READ_DATA);
    if (temp1 != AHTXX_ERROR) { 
        float humi1 = aht10.readHumidity(AHTXX_FORCE_READ_DATA);
        if (humi1 != AHTXX_ERROR) humi = humi1;
        temp = temp1-toffset;
        dataReady = true;
        DEBUG("-->[SLIB] AHT10 read\t\t: done!");
        unitRegister(UNIT::TEMP);
        unitRegister(UNIT::HUM);
    }
}
``` 
## Output:

```
10:33:22.834 > -->[WIFI] AP RSSI signal         : -75 dBm
10:33:22.838 > -->[HEAP] LOOP bytes used        : 1756b/052Kb
10:33:23.254 > -->[MQTT] mqtt.anaire.org        : connected!
10:33:27.726 > -->[SLIB] GCJA5 read             : done! 
10:33:27.730 > -->[SLIB] AM2320 read            : done! 
10:33:27.770 > -->[SLIB] SHT31 read             : done! 
10:33:27.774 > -->[SLIB] BMP280 read            : done! 
10:33:27.778 > -->[SLIB] Preview sensors values : PM1:3.0 PM2.5:4.0 PM10:5.0 T:28.3 H:58.4 P:1020.1 Alt:-57.1 
10:33:27.786 > -->[SLIB] Sensors devices count  : 4 (GCJA5,BMP280,AM232X,SHT31,)
10:33:27.794 > -->[SLIB] Sensors units count    : 7 (PM1,PM2.5,PM10,T,H,P,Alt,)
``` 

The alternative `Wire.setTimeout(50);` fails a lot, conflicts with `GCJA5` sensor and return FF on temperature from `BME680` sensor.

Thanks!

closed 2022-10-18 by hpsaturn

enjoyneering, 2022-10-18:
Good.

I also have [SCD30 library](SCD30_I2C](https://github.com/enjoyneering/SCD30_I2C) with tons of fetures. It has the same issue as AHTxx - default clock strech is 1ms.