enjoyneering/AHTxx #14: begin() is not returning value by "Wire.endTransmission()"

rob040, 2023-03-18:
In description of begin(), the return value is said to be "- returned value by Wire.endTransmission(): ", mentioning 5 different return values.
That is not the case. It returns bool true/false.

enjoyneering, 2023-03-19:
return values for Wire.endTransmission() for esp8266 can be found [here](https://github.com/esp8266/Arduino/blob/24c41524dc56c683d8926671bdd639d7411f2815/cores/esp8266/core_esp8266_si2c.cpp#L364)

return values for Wire.endTransmission() for AVR can be found [here](https://github.com/arduino/ArduinoCore-avr/blob/24e6edd475c287cdafee0a4db2eb98927ce3cf58/libraries/Wire/src/utility/twi.c#L253)

closed 2023-03-19 by enjoyneering

rob040, 2023-03-19:
Pretty strong defence and immediate closure of issue. Is that necessary?

> my begin() library is not Wire.endTransmission() and the description doesn't indicate that begin() will return 5 different results.

Well, in AHTxx.cpp, one can read:

/**************************************************************************/
/*
    begin()
    Initialize I2C & sensor
    NOTE:
    - call this function before doing anything else!!!
    - speed in Hz, stretch in usec
    - **returned value** by "Wire.endTransmission()":
      **- 0 success**
      **- 1 data too long to fit in transmit data buffer**
      **- 2 received NACK on transmit of address**
      **- 3 received NACK on transmit of data**
      **- 4 other error**
*/
/**************************************************************************/
#if defined (__AVR__)
bool AHTxx::begin(uint32_t speed, uint32_t stretch)