enjoyneering/AHTxx #15: mysensors

bachy, 2023-04-02:
hello, and thanks for the library.

It works smoothly with an arduino nano + aht10 board alone, but if i try to get it work in the context of a [mysensors](https://github.com/mysensors/MySensors) node, the i2c fails systematically to establish communication.
I tried to dive into the library and figured out that the `_address` is messed up (i guess by mysensors).
Actually if i replace in `ahtxx.cpp` all `_address` by `0x38` it works.
I'm a web developer and completely newbe in c++, could you please help me to solve this ?
Thanks

bachy, 2023-04-03:
hi,

i replaced _address by __i2cAHTAddress in ahtxx library and now it works together with mysensors.
what i don't understand is that _address is a private property of AHTxx class, so it should be encapsulated, how could mysensors overwrite it ?

ps: i only renamed _address in AHTxx and not in mysensors because mysensors is a way more complicated library :)

enjoyneering, 2023-04-03:
Hi,

'...figured out that the _address is messed up (i guess by mysensors).'

It is impossible, because is `_address`. The private data members cannot be accessed from outside the class (mysensors can not see it).

Сan you try my original library but with class initialization line

'''AHTxx aht10(0x38, AHT1x_SENSOR); //sensor address, sensor type'''

bachy, 2023-04-04:
ok, my bad, again i'm noob in cpp but that's what i understood by my self, _address is a private variable.
indeed, using ```AHTxx aht10(0x38 ...``` class initialisation, it works!

thank you for the answer ! and for the library !

closed 2023-04-04 by bachy

enjoyneering, 2023-04-06:
good. it looks like somewhere in the mysensors library there is a macro with the same name AHTXX_ADDRESS_X38 and mysensor **deliberately** overrides it, otherwise the compiler would throw an error