enjoyneering/AHTxx #17: Is there really a AHTXX_STATUS_REG command?

dquadros, 2023-06-04:
There is no mention to this command in the AHT10 datasheet. In the AHT20 datasheet, it is not listed in the commands, there is only the note "you can get a byte of status word by sending 0x71".  0x71 happens to be the address of the sensor plus the read bit. 

My take on this is that there is no need to send a command to get the status (and the readings), you just do an I2C read transaction. I tried this with an AHT10 and it worked. I do not have the other models, so cannot guarantee it will work for them.

keyurthumar, 2024-10-20:
Hello @dquadros 
I am not able to find the datasheet for AHT2415C00.
If you can help me with that, it would be great help for me.
Thank you

dquadros, 2024-10-20:
Sorry, I only found a "Product Summary" with no information on the I2C communication.
Daniel