enjoyneering/AHTxx #20: How i can use two sensor at same I2c port (0x38 & 0x39)?

miborbolla, 2023-10-11:
In my project i conect two AHT10 at diferent address (0x38 & 0x39)

How i can define, initialize and obtain temp and humidity of each sensor?

Y run the i2c scanner and can see the two sensor, but I no undersatnd how to use the library.

you can helpme?

miborbolla, 2023-10-11:
> In my project i conect two AHT10 at diferent address (0x38 & 0x39)
> 
> How i can define, initialize and obtain temp and humidity of each sensor?
> 
> Y run the i2c scanner and can see the two sensor, but I no undersatnd how to use the library.
> 
> you can helpme?

=========================================================

Here mi response:

`
/*

How to use two ATH10 sensor at 0x38 and 0x39 at same I2C port

*/

#include <Wire.h>
#include <AHTxx.h> 

float TempInt; //En esta variable se contener La temperatura interna del sensor AHT10 
float HumInt; //Vaiavle que contiene la humedad interna del sensor AHT10
float TempExt; //En esta variable se contener La temperatura externa del sensor AHT10 
float HumExt; //Vaiavle que contiene la humedad externa del sensor AHT10

AHTxx aht10_interno(0X38, AHT1x_SENSOR); //declaramos la direccion y tipo de sensor interno
AHTxx aht10_externo(0X39, AHT1x_SENSOR); //declaramos la direccion y tipo de sensor interno

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);

//inicializamos el sensor interno
while (aht10_interno.begin() != true)
  {
    Serial.println(F("Internal sensor not connected or fail to load calibration coefficient")); //(F()) save string to flash & keeps dynamic memory free
   }
  Serial.println(F("aht10_interno OK"));

//inicializamos el sensor externo
while (aht10_externo.begin() != true)
  {
    Serial.println(F("External sensor not connected or fail to load calibration coefficient")); //(F()) save string to flash & keeps dynamic memory free
   }
  Serial.println(F("aht10_externo OK"));
}

void SensorAHT10(){
//=================================sensor 1=========================  
  TempInt = aht10_interno.readTemperature(); //Obtenemos la temperatura Interna
  HumInt = aht10_interno.readHumidity(); //Obtenemos la humedad interna

Serial.print("Internal Temp=");Serial.print(TempInt);Serial.print("°C  Internal Humidity=");Serial.println(HumInt);
//=================================sensor 2=========================
  TempExt = aht10_externo.readTemperature(); //Obtenemos la temperatura Externa
  HumExt = aht10_externo.readHumidity(); //Obtenemos la humedad Externa

Serial.print("External Temp=");Serial.print(TempExt);Serial.print("°C  External Humidity=");Serial.println(HumExt);
}

void loop() {
  // put your main code here, to run repeatedly:
SensorAHT10();
delay(5000);

}

`

enjoyneering, 2023-10-16:
Hi,

To activate sensor address 0x39, you need to connect the AHT10 address pin to Vcc.

enjoyneering, 2023-11-29:
no responce

closed 2023-11-29 by enjoyneering